Haskell-Things/ImplicitCAD #389: How do I make and reuse parts?

FlyMyPG, 2021-05-06:
It may be implicit somewhere, but I found no explicit mention or example showing how to make a "thing" in a file, then reuse it both in that file and in other files.

For example, a library of standard bolts and nuts, screws, nails, etc.  Or a parametric rafter for making a roof on a house.  And so on.

If this capability exists, please expose it more prominently.

sorki, 2021-05-13:
Similar to OpenSCAD, implicits SCAD engine supports `include` (and `use`), see e.g. https://github.com/colah/ImplicitCAD/blob/master/Examples/example23.escad and you could possibly reuse some OpenSCAD libraries like MCAD (not everything is going to work due to hull). 

When using Haskell eDSL you just use Haskells import functionality.

thiskappaisgrey, 2021-06-15:
Hi, I have a related question, is there a way to import OpenScad libraries(or import stuff writen in `escad` when using the Haskell eDSL? For example, I want to use these parts: https://github.com/nophead/NopSCADlib in a `design.hs` file, is that possible?  I'm a haskell beginner, so an example would be helpful. Thank you so much!

> Similar to OpenSCAD, implicits SCAD engine supports `include` (and `use`), see e.g. https://github.com/colah/ImplicitCAD/blob/master/Examples/example23.escad and you could possibly reuse some [...]