Haskell-Things/ImplicitCAD #396: Install not working on Ubuntu 18.04

unrealwill, 2021-06-29:
Hello,
I tried following your install procedure : 

The latest release fail with : 
```
apt-get install ghc cabal-install
cabal update && cabal install implicit
```
Fails with : 
```
Downloading the latest package list from hackage.haskell.org
Skipping download: local and remote files match.
Resolving dependencies...
Configuring implicit-0.3.0.1...
Building implicit-0.3.0.1...
Failed to install implicit-0.3.0.1
Build log ( /home/username/.cabal/logs/implicit-0.3.0.1.log ):
cabal: Entering directory '/tmp/cabal-tmp-15816/implicit-0.3.0.1'
Configuring implicit-0.3.0.1...
Building implicit-0.3.0.1...
Preprocessing library implicit-0.3.0.1...

on the commandline: warning:
    unrecognised warning flag: -Wmissing-export-lists

on the commandline: warning:
    unrecognised warning flag: -Wmissing-home-modules
[ 1 of 48] Compiling Graphics.Implicit.FastIntUtil ( Graphics/Implicit/FastIntUtil.hs, dist/build/Graphics/Implicit/FastIntUtil.o )
[ 2 of 48] Compiling Graphics.Implicit.IntegralUtil ( Graphics/Implicit/IntegralUtil.hs, dist/build/Graphics/Implicit/IntegralUtil.o )
[ 3 of 48] Compiling Graphics.Implicit.ExtOpenScad.Parser.Lexer ( Graphics/Implicit/ExtOpenScad/Parser/Lexer.hs, dist/build/Graphics/Implicit/ExtOpenScad/Parser/Lexer.o )

./Graphics/Implicit/ExtOpenScad/Parser/Lexer.hs: warning:
    Could not specialise imported function ‘char’
    Probable fix: add INLINEABLE pragma on ‘char’
[ 4 of 48] Compiling Graphics.Implicit.Definitions ( Graphics/Implicit/Definitions.hs, dist/build/Graphics/Implicit/Definitions.o )

./Graphics/Implicit/Definitions.hs: warning:
    Could not specialise imported function ‘Data.Either.$w$cshowsPrec’
      when specialising ‘Data.Either.$fShowEither_$cshowsPrec’
    Probable fix: add INLINEABLE pragma on ‘Data.Either.$w$cshowsPrec’

./Graphics/Implicit/Definitions.hs: warning:
    Could not specialise imported function ‘GHC.Show.$fShowMaybe_$cshowsPrec’
    Probable fix: add INLINEABLE pragma on ‘GHC.Show.$fShowMaybe_$cshowsPrec’

./Graphics/Implicit/Definitions.hs: warning:
    Could not specialise imported function ‘GHC.Show.$w$cshowsPrec4’
      when specialising ‘GHC.Show.$fShow(,,)_$cshowsPrec’
    Probable fix: add INLINEABLE pragma on ‘GHC.Show.$w$cshowsPrec4’
[ 5 of 48] Compiling Graphics.Implicit.Export.MarchingSquaresFill ( Graphics/Implicit/Export/MarchingSquaresFill.hs, dist/build/Graphics/Implicit/Export/MarchingSquaresFill.o )

Graphics/Implicit/Export/MarchingSquaresFill.hs:31:9: warning: [-Wmonomorphism-restriction]
    • The Monomorphism Restriction applies to the bindings for ‘n’,
                                                               ‘nx’, ‘ny’
        Consider giving a type signature for these binders
    • In the expression:
        let
          d = p2 ^-^ p1
          nx :: ℕ
          ny :: ℕ
          ....
        in triangles
      In an equation for ‘getContourMesh’:
          getContourMesh p1 p2 res obj
            = let
                d = p2 ^-^ p1
                nx :: ℕ
                ....
              in triangles

./Graphics/Implicit/Export/MarchingSquaresFill.hs: warning:
    Could not specialise imported function ‘Data.Foldable.$fFoldable[]_$cfold’
    Probable fix: add INLINEABLE pragma on ‘Data.Foldable.$fFoldable[]_$cfold’

./Graphics/Implicit/Export/MarchingSquaresFill.hs: warning:
    Could not specialise imported function ‘GHC.Float.$w$cproperFraction’
      when specialising ‘GHC.Float.$fRealFracDouble_$cproperFraction’
      when specialising ‘GHC.Float.$fRealFracDouble_$cfloor’
    Probable fix: add INLINEABLE pragma on ‘GHC.Float.$w$cproperFraction’
[ 6 of 48] Compiling Graphics.Implicit.Export.RayTrace ( Graphics/Implicit/Export/RayTrace.hs, dist/build/Graphics/Implicit/Export/RayTrace.o )

./Graphics/Implicit/Export/RayTrace.hs: warning:
    Could not specialise imported function ‘Data.VectorSpace.$w$cp1InnerSpace1’
      when specialising ‘Data.VectorSpace.$fInnerSpace(,,)_$cp1InnerSpace’
    Probable fix: add INLINEABLE pragma on ‘Data.VectorSpace.$w$cp1InnerSpace1’

./Graphics/Implicit/Export/RayTrace.hs: warning:
    Could not specialise imported function ‘GHC.Real.$w$s$cround’
      when specialising ‘GHC.Real.$fRealFracRatio_$s$cround’
    Probable fix: add INLINEABLE pragma on ‘GHC.Real.$w$s$cround’

./Graphics/Implicit/Export/RayTrace.hs: warning:
    Could not specialise imported function ‘Data.VectorSpace.$w$c<.>1’
      when specialising ‘Data.VectorSpace.$fInnerSpace(,,)_$c<.>’
      when specialising ‘Data.VectorSpace.$fInnerSpace(,,)’
    Probable fix: add INLINEABLE pragma on ‘Data.VectorSpace.$w$c<.>1’

./Graphics/Implicit/Export/RayTrace.hs: warning:
    Could not specialise imported function ‘GHC.Show.$w$cshowsPrec4’
      when specialising ‘GHC.Show.$fShow(,,)_$cshowsPrec’
    Probable fix: add INLINEABLE pragma on ‘GHC.Show.$w$cshowsPrec4’
[ 7 of 48] Compiling Graphics.Implicit.Export.Render.Definitions ( Graphics/Implicit/Export/Render/Definitions.hs, dist/build/Graphics/Implicit/Export/Render/Definitions.o )
[ 8 of 48] Compiling Graphics.Implicit.Export.Render.GetLoops ( Graphics/Implicit/Export/Render/GetLoops.hs, dist/build/Graphics/Implicit/Export/Render/GetLoops.o )

Graphics/Implicit/Export/Render/GetLoops.hs:8:72: error:
    Module ‘Prelude’ does not export ‘(<>)’
cabal: Leaving directory '/tmp/cabal-tmp-15816/implicit-0.3.0.1'
cabal: Error: some packages failed to install:
implicit-0.3.0.1 failed during the building phase. The exception was:
ExitFailure 1
```

The development version fails when installing the dependencies : 
(It was complaining about a missing happy so I installed it manually, but now it fails with 
Failed to install plugins-1.6.2)

```
cabal update && cabal install hashmap parallel parsec plugins JuicyPixels blaze-builder blaze-markup blaze-svg storable-endian unordered-containers vector-space
Downloading the latest package list from hackage.haskell.org
Skipping download: local and remote files match.
Resolving dependencies...
Configuring plugins-1.6.2...
Building plugins-1.6.2...
Failed to install plugins-1.6.2
Build log ( /home/username/.cabal/logs/plugins-1.6.2.log ):
cabal: Entering directory '/tmp/cabal-tmp-16762/plugins-1.6.2'
Configuring plugins-1.6.2...
configure: WARNING: unrecognized options: --with-compiler
checking build system type... x86_64-unknown-linux-gnu
checking for ghc... ghc
checking for value of __GLASGOW_HASKELL__... 800
checking for ghc library directory... /usr/lib/ghc
checking for gcc... /usr/bin/gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /usr/bin/gcc accepts -g... yes
checking for /usr/bin/gcc option to accept ISO C89... none needed
checking for arc4random... no
checking for a BSD-compatible install... /usr/bin/install -c
configure: creating ./config.status
config.status: creating config.mk
config.status: creating testsuite/makewith/io/TestIO.conf
config.status: creating testsuite/makewith/unsafeio/Unsafe.conf
config.status: creating config.h
configure: WARNING: unrecognized options: --with-compiler
Building plugins-1.6.2...
Preprocessing library plugins-1.6.2...

src/System/Plugins/Consts.hs:24:2: error:
     fatal error: config.h: No such file or directory
     
      ^         
compilation terminated.
`gcc' failed in phase `C pre-processor'. (Exit code: 1)
cabal: Leaving directory '/tmp/cabal-tmp-16762/plugins-1.6.2'
cabal: Error: some packages failed to install:
plugins-1.6.2 failed during the building phase. The exception was:
ExitFailure 1
```
Can you advise ?
Thanks

tobbelobb, 2021-07-05:
I have the same problem (`cabal install implicit`) failing on Ubuntu 18.04.

I haven't tried the dev build.

sorki, 2022-04-23:
I've tested a build on ubuntu-18.04 using modified CI environment https://github.com/sorki/ImplicitCAD/commit/03822355f65ab71a4829d5eeee83ed9326bc8031 and it passed for all supported GHC versions with Cabal 3.4 https://github.com/sorki/ImplicitCAD/actions/runs/2212608265

Closing, re-open if it still doesn't work for you.

sorki, 2022-04-23:
To clarify, development version does build fine now. For `cabal instal implicit` we need to publish a new release.

unrealwill, 2022-04-23:
Thanks. Can't test anymore as I have moved to 20.04.

sorki, 2023-10-27:
Closing as the new version was published some time ago.

closed 2023-10-27 by sorki