Haskell-Things/ImplicitCAD #398: Doesn't build on Windows

dannypike, 2021-09-03:
I just tried setting this up on a Windows PC and it doesn't build:

```
C:\WINDOWS\system32>cabal update
Downloading the latest package list from hackage.haskell.org
Updated package list of hackage.haskell.org to the index-state 2021-09-02T22:10:56Z
To revert to previous state run:
    cabal v2-update 'hackage.haskell.org,2021-08-13T05:39:09Z'

C:\WINDOWS\system32>cabal install implicit
Resolving dependencies...
Build profile: -w ghc-9.0.1 -O1
In order, the following will be built (use -v for more details):
 - hashable-1.3.3.0 (lib) (requires download & build)
 - unordered-containers-0.2.14.0 (lib) (requires build)
 - implicit-0.0.5 (lib:implicit, exe:extopenscad) (requires build)
Downloading  hashable-1.3.3.0
Downloaded   hashable-1.3.3.0
Starting     hashable-1.3.3.0 (lib)
Building     hashable-1.3.3.0 (lib)
Installing   hashable-1.3.3.0 (lib)
Completed    hashable-1.3.3.0 (lib)
Starting     unordered-containers-0.2.14.0 (lib)
Building     unordered-containers-0.2.14.0 (lib)
Installing   unordered-containers-0.2.14.0 (lib)
Completed    unordered-containers-0.2.14.0 (lib)
Starting     implicit-0.0.5 (all, legacy fallback)
Building     implicit-0.0.5 (all, legacy fallback)

Failed to build implicit-0.0.5.
Build log (
D:\Users\Dan\AppData\Roaming\cabal\logs\ghc-9.0.1\implicit-0.0.5-af197750bd53fbd3cd1b0f7dbbf07049d5da9c3d.log
):
Preprocessing executable 'extopenscad' for implicit-0.0.5..
Building executable 'extopenscad' for implicit-0.0.5..

Graphics\Implicit\ExtOpenScad\Util\OVal.hs:1:14: warning:
    -XOverlappingInstances is deprecated: instead use per-instance pragmas OVERLAPPING/OVERLAPPABLE/OVERLAPS
  |
1 | {-# LANGUAGE OverlappingInstances, ViewPatterns, RankNTypes, ScopedTypeVariables, TypeSynonymInstances, FlexibleInst
ances #-}
  |              ^^^^^^^^^^^^^^^^^^^^

<no location info>: warning: [-Wmissing-home-modules]
    These modules are needed for compilation but not listed in your .cabal file's other-modules:
        Graphics.Implicit
        Graphics.Implicit.Definitions
        Graphics.Implicit.Export
        Graphics.Implicit.Export.Definitions
        Graphics.Implicit.Export.MarchingSquares
        Graphics.Implicit.Export.MarchingSquaresFill
        Graphics.Implicit.Export.NormedTriangleMeshFormats
        Graphics.Implicit.Export.PolylineFormats
        Graphics.Implicit.Export.RayTrace
        Graphics.Implicit.Export.Render
        Graphics.Implicit.Export.Render.Definitions
        Graphics.Implicit.Export.Render.GetLoops
        Graphics.Implicit.Export.Render.GetSegs
        Graphics.Implicit.Export.Render.HandlePolylines
        Graphics.Implicit.Export.Render.HandleSquares
        Graphics.Implicit.Export.Render.Interpolate
        Graphics.Implicit.Export.Render.RefineSegs
        Graphics.Implicit.Export.Render.TesselateLoops
        Graphics.Implicit.Export.Symbolic.Rebound2
        Graphics.Implicit.Export.Symbolic.Rebound3
        Graphics.Implicit.Export.SymbolicFormats
        Graphics.Implicit.Export.SymbolicObj2
        Graphics.Implicit.Export.SymbolicObj3
        Graphics.Implicit.Export.TextBuilderUtils
        Graphics.Implicit.Export.TriangleMeshFormats
        Graphics.Implicit.Export.Util
        Graphics.Implicit.ExtOpenScad
        Graphics.Implicit.ExtOpenScad.Default
        Graphics.Implicit.ExtOpenScad.Definitions
        Graphics.Implicit.ExtOpenScad.Eval.Expr
        Graphics.Implicit.ExtOpenScad.Eval.Statement
        Graphics.Implicit.ExtOpenScad.Parser.Expr
        Graphics.Implicit.ExtOpenScad.Parser.Statement
        Graphics.Implicit.ExtOpenScad.Parser.Util
        Graphics.Implicit.ExtOpenScad.Primitives
        Graphics.Implicit.ExtOpenScad.Util.ArgParser
        Graphics.Implicit.ExtOpenScad.Util.OVal
        Graphics.Implicit.ExtOpenScad.Util.StateC
        Graphics.Implicit.MathUtil
        Graphics.Implicit.ObjectUtil
        Graphics.Implicit.ObjectUtil.GetBox2
        Graphics.Implicit.ObjectUtil.GetBox3
        Graphics.Implicit.ObjectUtil.GetImplicit2
        Graphics.Implicit.ObjectUtil.GetImplicit3
        Graphics.Implicit.Primitives
[ 1 of 46] Compiling Graphics.Implicit.Definitions ( Graphics\Implicit\Definitions.hs, dist\build\extopenscad\extopensca
d-tmp\Graphics\Implicit\Definitions.o )
[ 2 of 46] Compiling Graphics.Implicit.Export.Definitions ( Graphics\Implicit\Export\Definitions.hs, dist\build\extopens
cad\extopenscad-tmp\Graphics\Implicit\Export\Definitions.o )
[ 3 of 46] Compiling Graphics.Implicit.Export.MarchingSquares ( Graphics\Implicit\Export\MarchingSquares.hs, dist\build\
extopenscad\extopenscad-tmp\Graphics\Implicit\Export\MarchingSquares.o )
[ 4 of 46] Compiling Graphics.Implicit.Export.MarchingSquaresFill ( Graphics\Implicit\Export\MarchingSquaresFill.hs, dis
t\build\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\MarchingSquaresFill.o )
[ 5 of 46] Compiling Graphics.Implicit.Export.Render.Definitions ( Graphics\Implicit\Export\Render\Definitions.hs, dist\
build\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Render\Definitions.o )
[ 6 of 46] Compiling Graphics.Implicit.Export.Render.GetLoops ( Graphics\Implicit\Export\Render\GetLoops.hs, dist\build\
extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Render\GetLoops.o )
[ 7 of 46] Compiling Graphics.Implicit.Export.Render.HandlePolylines ( Graphics\Implicit\Export\Render\HandlePolylines.h
s, dist\build\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Render\HandlePolylines.o )
[ 8 of 46] Compiling Graphics.Implicit.Export.Render.HandleSquares ( Graphics\Implicit\Export\Render\HandleSquares.hs, d
ist\build\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Render\HandleSquares.o )
[ 9 of 46] Compiling Graphics.Implicit.Export.Render.Interpolate ( Graphics\Implicit\Export\Render\Interpolate.hs, dist\
build\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Render\Interpolate.o )
[10 of 46] Compiling Graphics.Implicit.Export.Symbolic.Rebound2 ( Graphics\Implicit\Export\Symbolic\Rebound2.hs, dist\bu
ild\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Symbolic\Rebound2.o )
[11 of 46] Compiling Graphics.Implicit.Export.Symbolic.Rebound3 ( Graphics\Implicit\Export\Symbolic\Rebound3.hs, dist\bu
ild\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\Symbolic\Rebound3.o )
[12 of 46] Compiling Graphics.Implicit.Export.TextBuilderUtils ( Graphics\Implicit\Export\TextBuilderUtils.hs, dist\buil
d\extopenscad\extopenscad-tmp\Graphics\Implicit\Export\TextBuilderUtils.o )

Graphics\Implicit\Export\TextBuilderUtils.hs:20:6: error:
    Ambiguous occurrence `<>'
    It could refer to
       either `Prelude.<>',
              imported from `Prelude' at Graphics\Implicit\Export\TextBuilderUtils.hs:5:8-48
              (and originally defined in `GHC.Base')
           or `Graphics.Implicit.Export.TextBuilderUtils.<>',
              defined at Graphics\Implicit\Export\TextBuilderUtils.hs:72:1
   |
20 |     ,(<>)
   |      ^^^^
cabal.exe: Failed to build implicit-0.0.5. See the build log above for
details.

C:\WINDOWS\system32>
```

I had a problem with the Haskell installation. This PC has not seen Haskell before and, after completing the installation steps for Windows, running "GHC" from a command prompt didn't work. I had to search the drive to see where the Chocolatey install had put "ghc.exe" and add that directory to the PATH manually. So, maybe there's something else that I haven't done / failed to get set?

The install instructions for Windows don't seem to mention the need for any special configuration other than running Chocolatey to install cabal, stack and haskell-dev.

julialongtin, 2021-09-03:
something has selected a very old version of implicitcad to install. this version doesn't work with your (newer) ghc.

I'd work on determining WHY your system is looking at that old of a version. maybe one of the libraries we use is only available in an 'old' version on windows?

dannypike, 2021-09-04:
Many thanks for the speedy response.

Unfortunately, my total experience of using Haskell is in trying to get this working, so I'm not sure that I have the skills for verifying and fixing the build structure (I'm a C++/C# user). It seems like I should just wait for someone who knows what they're doing in cabal etc. to come across this problem and fix it.

sorki, 2022-04-02:
Some GHC9 issues were fixed as ecosystem progressed so this might work now. Mind re-trying the build using a clone of the repository and running `cabal build` inside its root?

dannypike, 2022-04-02:
@sorki Thank you.

Should I be cloning the latest GitHub versions of Haskell, too? I can't remember how I installed it last time, but it doesn't appear to be from GitHub (I can usually tell that from the path to the exe on my PC).

I have cabal version 3.4.0.0 and ghc version 9.0.1 at the moment.

sorki, 2022-04-02:
> Should I be cloning the latest GitHub versions of Haskell, too? I can't remember how I installed it last time, but it doesn't appear to be from GitHub (I can usually tell that from the path to the exe on my PC).
> 
> I have cabal version 3.4.0.0 and ghc version 9.0.1 at the moment.

No need to - this combination is quite close to what we now test ImplicitCAD with (Cabal 3.4 and GHC9.0.2) so it should be good.

dannypike, 2022-04-02:
OK. In that case, I'm afraid the build still does not work:

```
D:\work\github.forks\Haskell-Things>git clone https://github.com/Haskell-Things/ImplicitCAD.git
Cloning into 'ImplicitCAD'...
remote: Enumerating objects: 12341, done.
remote: Counting objects: 100% (562/562), done.
remote: Compressing objects: 100% (352/352), done.
Receiving objects: 100% (12341/12341), 5.46 MiB | 4.00 MiB/s, done.d 11779

Resolving deltas: 100% (7518/7518), done.

D:\work\github.forks\Haskell-Things>cd ImplicitCAD

D:\work\github.forks\Haskell-Things\ImplicitCAD>cabal build
Warning: The package list for 'hackage.haskell.org' is 211 days old.
Run 'cabal update' to get the latest list of available packages.
Resolving dependencies...
Build profile: -w ghc-9.0.1 -O1
In order, the following will be built (use -v for more details):
 - StateVar-1.2.2 (lib) (requires download & build)
 - base-orphans-0.8.5 (lib) (requires download & build)
 - binary-orphans-1.0.1 (lib) (requires download & build)
 - blaze-builder-0.4.2.1 (lib) (requires build)
 - byteorder-1.0.4 (lib:byteorder) (requires build)
 - call-stack-0.4.0 (lib) (requires download & build)
 - cereal-0.5.8.1 (lib) (requires download & build)
 - clock-0.8.2 (lib) (requires download & build)
 - colour-2.3.6 (lib) (requires build)
 - hashable-1.3.3.0 (lib) (requires build)
 - hspec-discover-2.8.3 (lib) (requires download & build)
 - indexed-traversable-0.1.1 (lib) (requires download & build)
 - integer-logarithms-1.0.3.1 (lib) (requires download & build)
 - mintty-0.1.2 (lib) (requires build)
 - parallel-3.2.2.0 (lib) (requires build)
 - primitive-0.7.2.0 (lib) (requires build)
 - reflection-2.1.6 (lib) (requires download & build)
 - semigroups-0.19.2 (lib) (requires download & build)
 - setenv-0.1.1.3 (lib) (requires download & build)
 - show-combinators-0.2.0.0 (lib) (requires download & build)
 - splitmix-0.1.0.3 (lib) (requires download & build)
 - tagged-0.8.6.1 (lib) (requires download & build)
 - th-abstraction-0.4.3.0 (lib) (requires download & build)
 - transformers-compat-0.7 (lib) (requires build)
 - void-0.7.3 (lib) (requires download & build)
 - zlib-0.6.2.3 (lib) (requires build)
 - contravariant-1.5.5 (lib) (requires download & build)
 - blaze-markup-0.8.2.8 (lib) (requires build)
 - storable-endian-0.2.6 (lib:storable-endian) (requires build)
 - HUnit-1.6.2.0 (lib) (requires download & build)
 - unordered-containers-0.2.14.0 (lib) (requires build)
 - ansi-terminal-0.11 (lib) (requires build)
 - vector-0.12.3.0 (lib) (requires build)
 - scientific-0.3.7.0 (lib) (requires download & build)
 - random-1.2.0 (lib) (requires download & build)
 - distributive-0.6.2.1 (lib) (requires download & build)
 - transformers-base-0.4.6 (lib) (requires download & build)
 - blaze-svg-0.3.6.1 (lib) (requires build)
 - hspec-expectations-0.8.2 (lib) (requires download & build)
 - ansi-wl-pprint-0.6.9 (lib) (requires build)
 - indexed-traversable-instances-0.1 (lib) (requires download & build)
 - JuicyPixels-3.3.5 (lib) (requires build)
 - bytes-0.17.1 (lib) (requires download & build)
 - tf-random-0.5 (lib) (requires download & build)
 - QuickCheck-2.14.2 (lib) (requires download & build)
 - comonad-5.0.8 (lib) (requires download & build)
 - optparse-applicative-0.16.1.0 (lib) (requires build)
 - quickcheck-io-0.2.0 (lib) (requires download & build)
 - bifunctors-5.5.11 (lib) (requires download & build)
 - hspec-core-2.8.3 (lib) (requires download & build)
[... 126 lines cut ...]
Downloading  hspec-expectations-0.8.2
Downloaded   hspec-expectations-0.8.2
Downloading  quickcheck-io-0.2.0
Downloaded   quickcheck-io-0.2.0
Downloading  hspec-core-2.8.3
Downloaded   hspec-core-2.8.3
Downloading  hspec-2.8.3
Installing   cereal-0.5.8.1 (lib)
Completed    cereal-0.5.8.1 (lib)

Failed to build byteorder-1.0.4. The failure occurred during the configure
step.
Build log (
D:\Users\peter\AppData\Roaming\cabal\logs\ghc-9.0.1\byteorder-1.0.4-6cc448b2d867a6e8100445a166d24e8e674aaadb.log
):
Configuring byteorder-1.0.4...
D:\Users\peter\AppData\Local\Temp\ghc7869.o.tmp: renameFile:renamePath:MoveFileEx "\\\\?\\D:\\Users\\peter\\AppData\\Local\\Temp\\ghc7869.o.tmp" Just "\\\\?\\D:\\Users\\peter\\AppData\\Local\\Temp\\ghc7869.o": does not exist (The system cannot find the file specified.)
cabal.exe: Failed to build byteorder-1.0.4 (which is required by
exe:extopenscad from implicit-0.3.0.1 and exe:docgen from implicit-0.3.0.1).
See the build log above for details.

D:\work\github.forks\Haskell-Things\ImplicitCAD>code D:\Users\peter\AppData\Roaming\cabal\logs\ghc-9.0.1\byteorder-1.0.4-6cc448b2d867a6e8100445a166d24e8e674aaadb.log

D:\work\github.forks\Haskell-Things\ImplicitCAD>
```

dannypike, 2022-04-02:
and that log file contains this:
```
Configuring byteorder-1.0.4...
D:\Users\peter\AppData\Local\Temp\ghc7869.o.tmp: renameFile:renamePath:MoveFileEx "\\\\?\\D:\\Users\\peter\\AppData\\Local\\Temp\\ghc7869.o.tmp" Just "\\\\?\\D:\\Users\\peter\\AppData\\Local\\Temp\\ghc7869.o": does not exist (The system cannot find the file specified.)
```

dannypike, 2022-04-02:
There is a file called: ```D:\Users\peter\AppData\Local\Temp\ghc7869.tmp```, so is there a pathname-construction issue in cabal when working in Windows?

sorki, 2022-04-02:
Thanks, interesting, this looks like an issue with `byteorder` having troubles with handling paths on Windows.

`byteorder` is indirectly pulled in by `storable-endian` package and it looks like we use only a small part of it which might be fixable so we don't depend on these two.

sorki, 2022-04-02:
Digging deeper it is actually caused by `byteorder` - the hint I found in your logs is `legacy fallback` which as explained in https://stackoverflow.com/questions/68173444/what-does-legacy-fallback-mean-when-cabal-is-building-packages , specifically `The cabal-version is less than 1.8` where `byteorder` uses `Cabal-Version:  >= 1.6`.

So we either need to get rid of the package itself or bump its cabal version. I will try to contact the maintainer since the package seems to be used widely.

dannypike, 2022-04-08:
Have you had a response from the maintainer of ```byteorder```?

Is there a ticket that I can track?

sorki, 2022-04-09:
I found a way around in #411.

sorki, 2022-04-23:
If anyone stumbles on this what helped was limiting parallelism to single core as the errors are caused by some sort of race-condition in Cabal - i.e. use `cabal build -j1`.

commit 3f7c13e7 (2022-04-23): binaryStl: replace float32LE with floatLE from Data.ByteString.Builder

This allows us to get rid of `storable-endian` dependency
and transitively problematic `byteorder`.

Closes #398.
closed 2022-04-23 by julialongtin