Haskell-Things/ImplicitCAD #417: Performance of `cube` depends heavily on dimensions

mokshasoft, 2022-05-02:
The following two calls take very different times to finish:
```
cube False (V3 200 95 45)
cube False (V3 4000 95 45) -- this is 32x slower
```

Code that triggers:
```
someFunc :: IO ()
someFunc = do
  putStrLn "Generating model1.stl fast"
  timeIt $ writeSTL 1 "model1.stl" $ cube False (V3 200 95 45)
  putStrLn "Generating model2.stl slow"
  timeIt $ writeSTL 1 "model2.stl" $ cube False (V3 4000 95 45)
  putStrLn "Done"
```

Trigger using https://github.com/mokshasoft/hus/tree/perf-error-1/Elements
```
$ stack build
...
$ stack exec Elements
Generating model1.stl fast
CPU time:  15.98s
Generating model2.stl slow
CPU time: 524.81s
Done
```

sorki, 2022-05-02:
The larger cube increases the volume to be evaluated significantly so this is currently expected. The cube results in implicit function which is evaluated along full range of the objects bounds.

mokshasoft, 2022-05-03:
Thanks! When I rearranged the code a bit I understand that it is the rendering that takes time (because of the rendering resolution) not the cube call. Changing the resolution to 20 makes the rendering almost instant.
```
c2 <- timeIt $ pure $ cube False (V3 4000 95 45)
timeIt $ writeSTL 20 "model2.stl" c2
```

closed 2022-05-03 by mokshasoft