Haskell-Things/ImplicitCAD #418: cube False (V3 3 0.095 0.045) produces no geometry

mokshasoft, 2022-05-02:
This code:
```
someFunc :: IO ()
someFunc = do
  putStrLn "Generating model3.stl"
  writeSTL 1 "model3.stl" $ cube False (V3 3 0.095 0.045)
  putStrLn "Done"
```
produces this "empty" stl-file:
```
solid ImplictCADExport
endsolid ImplictCADExport
```
Trigger using https://github.com/mokshasoft/hus/tree/cube-no-output/Elements
```
$ stack build
...
$ stack exec Elements
Generating model3.stl
Done
```

sorki, 2022-05-02:
Hi! You're asking the engine to render at resolution `1` (the first parameter to `writeSTL`) but your object is smaller than this. The resolution needs to be less than the minimum size of the feature you want to render (in your case something like `0.045` and lower should work).

mokshasoft, 2022-05-03:
Oh, a long time ago I tried to find out what that parameter were doing, but couldn't find it. Now I see that it's there in the docs. Thanks!

closed 2022-05-03 by mokshasoft