Haskell-Things/ImplicitCAD #442: cabal install error

oldbrad, 2023-08-04:
$ cd ~/git/ImplicitCAD
$ $ cabal install implicit
Resolving dependencies...
cabal: internal error when reading package index: failed to parse .cabal
fileThe package index or index cache is probably corrupt. Running cabal update
might fix it.
 ~ 0 ~
I have installed cabal-install from scratch and run cabal update successfully. I have tried:
$ rm -fr ~/.cabal/packages/hackage.haskell.org
$ cabal update; cabal install implicit
Downloading the latest package list from hackage.haskell.org
Resolving dependencies...
cabal: internal error when reading package index: failed to parse .cabal
fileThe package index or index cache is probably corrupt. Running cabal update might fix it.

I get no response from hackage-admin@haskell.org
I think I have a cabal version problem? cabal --version says:
cabal-install version 1.22.6.0
using version 1.22.5.0 of the Cabal library
Can anyone here help me fix the problem? How do I install latest cabal without a working cabal??

oldbrad, 2023-08-04:
OK Solved this myself . Classically the error message is erroneous! The problem is best described as 'cabal HELL' .. ha ha
Solution: Must have latest versions.  The best way is to use ghcup - here: https://www.haskell.org/ghcup/install/
Delete your package manager versions of ghc and cabal-install and delete ~/.cabal first.

julialongtin, 2023-08-21:
Haskell for the win? :)

closed 2024-01-06 by julialongtin