Haskell-Things/ImplicitCAD #443: ExtOpenScad Module Import not in scope

kyletreece, 2023-09-09:
Hi there! 

I got the error in the title of this issue when trying to run extopenscad for an scad script that includes a call to the import function, specifically to import in an SVG file. Seems like the openscad import function is not supported in implicitCAD. Is this true, or is there some other way to bring in SVG files?

Thanks in advance for your help!

sorki, 2023-10-22:
Hi!

> Seems like the openscad import function is not supported in implicitCAD. Is this true, or is there some other way to bring in SVG files?

Yes this is true. We currently don't have support for importing SVG/STL.

Reminds me of the previous attempt to make text rendering working by @isovector at https://github.com/isovector/cad-models/blob/master/src/Text.hs so maybe if you convert SVG into `polygon` this could work although there might a better way that would avoid direct sampling and instead treated SVG as a first class citizen that would only be sampled during rendering.

kyletreece, 2023-10-22:
Ok got it, thank you for the reply!

sorki, 2023-10-27:
I've altered the title a bit and gonna keep this open as a reminder for us. Also labeled as _Good first issue_.

googleson78, 2024-12-02:
> although there might a better way that would avoid direct sampling and instead treated SVG as a first class citizen that would only be sampled during rendering.

Would that include just building SVG into the `SymbolicObj2` as a primitive? Also, do you have some way to render SVG at that point in mind? It seems like quite a complicated task, unless there's some lib we can reuse.