Haskell-Things/ImplicitCAD #480: openscad quasiquote

aavogt, 2024-01-20:
I like `circle($fn=6)`, but I don't see it in [the haskell api](https://hackage.haskell.org/package/implicit-0.4.1.0/docs/Graphics-Implicit-Primitives.html#v:circle). While it is possible to encode such optional arguments in haskell, I think a quasiquoter would be better.

```haskell
hex1 d = [scad| circle(d, $fn=6) |] -- too light
hex2 d = [scad2d| circle($$d, $fn=6) |] -- too heavy
hex3 d = [scad| circle($(d), $fn=6) |] -- just right?
```

ovitus, 2024-08-25:
Second this. How can this be done via Haskell?

aavogt, 2024-08-25:
https://github.com/Haskell-Things/ImplicitCAD/blob/master/programs/extopenscad.hs#L168 shows how it parses scad/runs scad files. SymbolicObj3 contains functions, so the QuasiQuoter probably has to run the parser at runtime because there's no Lift instance, ie. it probably will look like:

```haskell
-- | QuasiQuoter for extopenscad expressions
-- 
-- > hex ::  Double -> IO SymbolicObj2
-- > hex d = [scad2d| circle($(d), $fn=6) |]
scad2d = QuasiQuoter { quoteExp = \ str -> do
  checkSyntax str -- optional: probably calls runOpenscad
  [| (\(_,obj2s,_obj3s,_) -> obj2s) <$> runOpenscad scadOpts _ $(antiquote str) |] }

-- | > antiquote "ab $(c) def" == [| "ab " ++ show c ++ " def" |]
antiquote :: String -> ExpQ
antiquote = _
```
But maybe the result doesn't have to be in IO.